V1EngineeringInc/MarlinBuilder #17: Formatting

V1EngineeringInc, 2020-09-03:
First a little change that needs to be made.

From Disable
EVENT_GCODE_SD_STOP 
is now
EVENT_GCODE_SD_ABORT

I think most of the settings in the CNC config are actually universal, that previous command, eeprom, arcs, ect. Should they stay separate or should I start moving them to v1 base config?

jeffeb3, 2020-09-03:
I don't understand. You're saying there are configs in the cnc config that should apply to all the machines (zxy/mp3dp included)?

I think the original decision for what went in there was based on what you had in your allted github. So if you want to change it, you should. We can review it, but I'm guessing that all makes sense.

V1EngineeringInc, 2020-09-03:
Yeah.

So right now cnc-config has /disable "sd stop". but I think every build should have that. So should I move it to v1 base config?

If that is yes I think turning on the eeprom should also be there as well as the arc settings.

-For whatever reason EVENT_GCODE_SD_STOP is now called EVENT_GCODE_SD_ABORT. So the current current release builds will try to home if you cancel a job from the LCD???? All bad! At least the Rambo config I just checked.

jeffeb3, 2020-09-03:
OK. That makes sense. Do you want me to PR that?

jeffeb3, 2020-09-03:
If we fix #18, that will at least alert us to problems like this.

V1EngineeringInc, 2020-09-03:
submitted #19

closed 2020-09-16 by V1EngineeringInc