V1EngineeringInc/MarlinBuilder #20: Check that all the readmes make sense

jeffeb3, 2020-09-03:
We just need to go through each config file and make sure the readmes are correct and complete. I added this step a bit late, and I am sure I am missing some.

Overnuts, 2021-01-05:
At the end of  https://github.com/V1EngineeringInc/MarlinBuilder/blob/main/src/docs/README.md
there are links to howto local build on  "linux" and "windows",  both links goes to the ubuntu one

jeffeb3, 2021-01-05:
That is intentional. The WSL is the same. 

This Issue is about the readmes insude tge release zips.

@vic10us.

kubark42, 2021-03-29:
> That is intentional. The WSL is the same.

I had the same issue. It looks like a typo, although your explanation resolves the question.

I wonder if this can be cleared up? It's strange to click on a link for windows documentation, which sends the reader to a file called `UBUNTU_BUILD.md`.

V1EngineeringInc, 2021-03-29:
I think #74 might clear things up?

kubark42, 2021-03-29:
Perhaps, although I think it's fundamentally an unfortunate filename. It reminds me of those brain teasers where you have to speak the color of the letters, but the letters spell the wrong color:

[image]