V1EngineeringInc/MarlinBuilder #21: Front Page to MarlinBuilds

jeffeb3, 2020-09-03:
We've talked about this before, but it would be nice to have a githubio page or a page in the docs that points to the right zip files.

V1EngineeringInc, 2020-09-03:
Getting there. I have not tested an actual CNC machine yet. I n testing on an actual MP3DP I found the eeprom and home on abort issue.

It is best to have a few real tests before we switch over. We would need another release as the those two things have changed as well as the 32nd stepping.

V1EngineeringInc, 2020-09-03:
I think it is easier to try and keep up with you two and work through the current PR's while we get this thing dialed in before we also add a bunch of public release issues at the same time.

I know we will always have PR's but since I am just learning this side we are merging at least one a day.

Maybe I should bump the firmware post asking for testers.

V1EngineeringInc, 2020-11-02:
I think this is good, all the firmware links point here now.

closed 2020-11-02 by V1EngineeringInc