V1EngineeringInc/MarlinBuilder #26: LCD Timout not working?

V1EngineeringInc, 2020-09-18:
Using the main branch, The LCD never returns to the main menu.

I have it set to 3 minutes, in the current main branch. Config_adv line 1095-6. At least on the mp3dp mini rambo version. I can try more boards a bit later this afternoon.

V1EngineeringInc, 2020-09-18:
Works now?

closed 2020-09-18 by V1EngineeringInc

jeffeb3, 2020-09-18:
Did you change something? I didn't see a PR. I'm worried you made a change, but it didn't get into main for some reason.

V1EngineeringInc, 2020-09-18:
No I narrowed it down. The timeout only works when it is not "printing". I assume that is a marlin bug. 

I just poured more coffee and was going to look into the main marlin PR's.

V1EngineeringInc, 2020-09-18:
https://github.com/MarlinFirmware/Marlin/pull/17887

Only thing I am seeing, but don't understand it enough to know if this caused the issue. Or I am missing something. In config-adv there is a setting in the custom script menu,   //#define USER_SCRIPT_RETURN  // Return to status screen after a script, not sure if maybe this needs to be set since we user some custom scripts. The issue happens if I just start a print and adjust the temp, it never goes back to the status screen, so it is not just custom scripts.