V1EngineeringInc/MarlinBuilder #40: Config for SKR 1.4 Turbo

EynErgy, 2020-10-30:
Hi,
I would like to use an SKR 1.4 Turbo board for this project.
I may be able to do it myself but I think the only change from a pro to 1.4 turbo is the chip name.
LP1769 instead of LPC1763

EynErgy, 2020-10-30:
HI,
Ok it builds the standard version but not the dual end stops versions.

I found out it is a bad definition of pinfolds for LPX1768 in Marlin.
But how to change that in MarlinBuilder??

I need this I think:
// Limit Switches
//
#define X_MIN_PIN P1_29
#define X_MAX_PIN P1_26
#define Y_MIN_PIN P1_28
#define Y_MAX_PIN P1_25
#define Z_MIN_PIN P1_27
#define Z_MAX_PIN P1_00

EynErgy, 2020-10-30:
Hi,
Ok I found out a way, not perfect but now it builds.
I ill check on a board as soon as I receive mine

V1EngineeringInc, 2020-10-30:
You should be able to do it by adding a config like this, https://github.com/V1EngineeringInc/MarlinBuilder/blob/main/src/configs/V1CNC_Skr1p3_2209

At least something similar, if you need to define pins that have not been defined you should probably submit that to the main Marlin to complete that board.

EynErgy, 2020-10-30:
Thanks. I have done it
Config for 1.4 and 1.4 Turbo made

I will test my pins file once I have a board and once ok I will submit it to the main marlin repo

I will keep you informed

jeffeb3, 2020-10-30:
The scripts we have can't affect the pins files. We have a few options:

1. Submit the change to the MarlinFirmware/Marlin repo. This will then get used here when they release a new version. But you have to convince them it is the right change.
2. Make our scripts work better. I need to rework the scripts anyway to combat trying to set options that are spelled wrong. We could add the ability to affect pins files. It will add some complexity (which adds maintenance costs in the long run).
3. Put the defines at the bottom of the config and then roll the dice. I don't know which files get read first. My guess is the configuration.h is read first. So this may not work. But if we just plop those defines at the end of the config, maybe they would stick around long enough to be picked up after the pins file. IDK.

In general, I'm ok with trying to support the turbo. I tried messing with it, but I bricked my board. I haven't had time to see if it is repairable. It won't be as reliable as the builds for the skr pro, but it should be a good startung point, and we can accumulate workarounds and "gotchas" here.

EynErgy, 2020-10-30:
I got a build ok. So as soon as I get my board I will test.
I will keep you informed of the results

EynErgy, 2020-11-03:
Ok got the board, connected and flashed firmwares.
The normal firmware seems ok, no errors , sappers moves.

The dual end stops firmware is strange. Steppers moves correctly but the screen complains of unknown command M221.

Do you know what happens here?

jeffeb3, 2020-11-03:
That is because the tft firmware is asking what the current flow rate is, but this furmware is configured to not use any extruder.

EynErgy, 2020-11-03:
That's what I thought. Ok i will have to check the config of the tft

tca72, 2021-02-26:
Hello
did you find out the solution for the M221. 
Do you have also configured with Dual Mechanical endstop ?

EynErgy, 2021-02-26:
Hi, I just changed to the basic mode and not the graphical mode. 
In basic it did not complain. But I wil try to figure out where in the code this check is done. 
The dual end stop works perfectly on my setup. The machine is ready and already working

tca72, 2021-02-26:
Great,

I have an issue on skr v1.4 the endstop on e1det and pwrdet  are not
producing any effect.
I've got same issue as you with TFT m221... I solve it by using marlin
instead of but fw, and also leaving extruder =1.

EynErgy, 2021-02-27:
Have you tried the firmware compiled here for endstops?

EynErgy, 2021-02-27:
The second x stop must be in e0det.  The second y stop in e1det. There is nothing in pwrdet

tca72, 2021-02-27:
Hello

There is no configuration available for the skr v1.4 on v1 engineering or
or skr v1.4 turbo. Or I did not find it.

EynErgy, 2021-02-27:
Hi, go to my fork of the repository on https://github.com/EynErgy/MarlinBuilder
Theer is a build for v1.4 and v1.4 Turbo.
The turbo version is the one I use. 
Just download the zip file, uncompress and get the package for your board

tca72, 2021-02-27:
Hello

I was facing many issues, most of them I found a solution, except for the
Endstop of Y1, Z2.

I download the latest version of code from BTT SKRV1.4 (
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3) , and apply the
changes of the V1CNC_Skr1p4_DualLR_2209
<https://github.com/EynErgy/MarlinBuilder/blob/main/src/configs/V1CNC_Skr1p4_DualLR_2209>
 configuration.
I my case I use TMC5160 instead of TMC2209 so I made those changes. I have
StepperOnline motors. I use a SKRV1.4, TMC5160, StepperOnline, and TFT 70.

But I think it miss something in your configuration files.
Because from what I think , Y2 stepper on E0M, Z2 stepper is on E1M.
That means that somes changes need to be done on pins_BTT_SKR_V1_4.h in
order to remap E0 pins to Y2  and E1 pins to Z2.

#define Y2_STEP_PIN                        P2_13
#define Y2_DIR_PIN                         P0_11
#define Y2_ENABLE_PIN                      P2_12
#ifndef Y2_CS_PIN
  #define Y2_CS_PIN                        P1_04
#endif

#define Z2_STEP_PIN                        P1_15
#define Z2_DIR_PIN                         P1_14
#define Z2_ENABLE_PIN                      P1_16
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                        P1_01
#endif

/*
#define E0_STEP_PIN                        P2_13
#define E0_DIR_PIN                         P0_11
#define E0_ENABLE_PIN                      P2_12
#ifndef E0_CS_PIN
  #define E0_CS_PIN                        P1_04
#endif

#define E1_STEP_PIN                        P1_15
#define E1_DIR_PIN                         P1_14
#define E1_ENABLE_PIN                      P1_16
#ifndef E1_CS_PIN
  #define E1_CS_PIN                        P1_01
#endif
*/

Also on SKRV1.4  the XMAX, YMAX and ZMAX are not existing, so we need to
define the endstop to E01DET, E1DET, PWRDET  :
So 2 solutions:
Solution 1: remap Y_MAX to PWRDET in pins_BTT_SKR_V1_4.h

//
// Limit Switches
//
#ifdef X_STALL_SENSITIVITY
  #define X_STOP_PIN                  X_DIAG_PIN
  #if X_HOME_DIR < 0
    #define X_MAX_PIN                      P1_26  // E0DET
  #else
    #define X_MIN_PIN                      P1_26  // E0DET
  #endif
// tca add {
#elif ENABLED(X_DUAL_ENDSTOPS)
  #ifndef X_MIN_PIN
    #define X_MIN_PIN                      P1_29  // X-STOP
  #endif
  #ifndef X_MAX_PIN
    #define X_MAX_PIN                      P1_26  // E0DET
[... 216 lines cut ...]

Also add :

#define Y2_MIN_POS 0 // lowrider add as Y2 is replacing E0 declaration
#define Z2_MIN_POS 0 // lowrider as Z2 is replacing E1 declaration
#define Y2_MAX_POS Y_BED_SIZE
 // lowrider add as Y2 is replacinig E0 declaration
#define Z2_MAX_POS 225 // lowrider ass as Z2 is replacing E1 declaration

Compilation ended with such changes.

And if we use a BigTreeTech TFT 70 or TFT43, we also need to
// #define AUTO_REPORT_TEMPERATURES  // TCA comment out, otherwise it
perform M155 S3 on display

Also I disable the #define SQUARE_WAVE_STEPPING , as this is not working
with TMC5160 or StepperOnline motor, provocates a strange noise and the
motor is not moving.

So I compile, and test, but still this version is not working ( Endstop of
Y2( E1DET , Z2 PWRDET are not taken into account, and then stop nothing).

If you have any clue?

BR

EynErgy, 2021-02-27:
Oh, it's a Low Rider you made. I was more focussed on the MPCNC on my side.
The pins layout is not the same effectively. And you need a ZMAX port, so effectively the PwrDet is good to use.
Have you tried the M119 gcode and checked the triggers are functional?
I had some issues with my plugs in fact, and I still have to replace them with better ones because I had a lot of false contacts.

tca72, 2021-02-27:
Hello

Ah oik, so probably some differences.
It is a Lowrider2 that I try to make ;-) , not easy task as this board
seems to be a little bit different than the SKRV1.3 ( this one has 6
endstops). This is a month that I try to solve this problem with endstop.
But then I plug the reprap endstop to PWRDET and E1DET, nothing changes,
but on X-STOP,n Y-STOP, Z-STOP ( meaning ports qualified as endstop) they
are working.

https://www.biqu.equipment/products/3d-printer-parts-mechanical-endstop-without-with-the-wheel-for-reprap-ramps-1-4-3d-printer-with-independent-packing?_pos=4&_sid=2db1aff0d&_ss=r&variant=12927115984994

In your MPCNC, do you have 5 endstops or 3 endstops?
I mean do yu use an endstop on X-STOP,n Y-STOP, Z-STOP only , or do you
also use the PWRDET or E1DET, E0DET    ?

best regards

EynErgy, 2021-02-27:
Hi,
On my MPCNC I use 4 end stops for the steppers (X1, X2, Y1, Y2) plus a plate for the Z
on the SKR 1,4 Turbo there is 6 positions that can be used as end stops, so technically since the non turbo board is identical you have enough pins.

Let me 2-3 days to look the docs and dive into the code to see if I can have a working build for you

tca72, 2021-02-27:
Hello

Thank you for your help.

Yes I was supposing also I can use the 6 endstop connectors,  even if only
3 are identified as it and 5 have same hardware ( resistors).

If you want,  this is my configuration:
https://drive.google.com/file/d/1K9ehY8LSt6l3Ca0zvHa-BTizaD163DVn/view?usp=sharing

BR.

jeffeb3, 2021-02-28:
There are turbo configs in main, but we haven't released them yet. There is a dual lr version too:

https://github.com/V1EngineeringInc/MarlinBuilder/actions/runs/532362610

Let me know if it works, and ask follow up questions in the forums.

EynErgy, 2021-03-01:
Thanks jeffeb3 but I made a fork of the repo to generate builds for 1,4 and 1,4 turbo.
The turbo version works perfectly for mpcnc with or without dual endstops.

tca72, what changes have you made specificaly for tmc5160 and steppersOnline?
I think I have all the rest ready.

tca72, 2021-03-02:
Hello

Right now, I've got an issue with the endstops of PWRDET (YMMAX) and E1DET
( ZMAX) that are not producing any result when pressed.
https://www.biqu.equipment/products/3d-printer-parts-mechanical-endstop-without-with-the-wheel-for-reprap-ramps-1-4-3d-printer-with-independent-packing?_pos=4&_sid=e342db26c&_ss=r&variant=12927115984994

Once solved I will communicate with you , to send you my configuration
changes.
Right now the changes with your configuration for CNC SKRV1_4 are:

configuration.h :
#define USE_XMIN_PLUG // Lowider manual endstop axe X
#define USE_YMIN_PLUG // Lowider manual endstop axe Y
#define USE_ZMIN_PLUG // Lowider manual endstop axe Z
#define USE_XMAX_PLUG
 // SKRV1.4 X+ does not exist, but Marlin 2.0X remaps it in Pins.h to
to E0DET pin 1.26
#define USE_YMAX_PLUG
 // SKRV1.4 Y+ does not exist, but Marlin 2.0X remaps it in Pins.h to
to E1DET pin 1.25
#define USE_ZMAX_PLUG
 // SKRV1.4 Z+ does not exist, but Marlin 2.0X remaps it in Pins.h to
to PWRDET pin 1.0
...
#define X_MIN_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true
 // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING false
 // Set to true to invert the logic of the probe.
... ...
#define X_DRIVER_TYPE  TMC5160
 // lowrider plug X to X stepper driver on SKR1.4
#define Y_DRIVER_TYPE  TMC5160
 // lowrider plug Y1 to Y stepper driver  on SKRV1.4
#define Z_DRIVER_TYPE  TMC5160
 // lowrider plug Z1 to ZMA & ZMB stepper driver on SKRV1.4, only ZMA
will be used ( as I want one stepper per motor, later on probably it
will be connected to ZBM if I use the resume on printing module)
//#define X2_DRIVER_TYPE A4988
#define Y2_DRIVER_TYPE TMC5160
 // lowrider plug Y2 to E1 stepper driver on SKRV1.4, so need to
change pins.h to E1
#define Z2_DRIVER_TYPE TMC5160
 // lowrider plug Z2 to ZMB stepper driver on SKRV1.4
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
//#define E0_DRIVER_TYPE A4988
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

In configuration_adv.h:
RSENSE need to be changed to 0.075, on each steepers position
wheren TMC5160 is installed:
#define ENDSTOPS_ALWAYS_ON_DEFAULT
...
#define Y_DUAL_STEPPER_DRIVERS
[... 225 lines cut ...]
#ifndef E1_CS_PIN
  #define E1_CS_PIN                        -1
#endif

// E0 becomes Y2 :
// Lowrider if EXTRUDERS = 0 , so marlin will E0=>X1 et E1=>Y2 but
Lowrider2 I want E0=Y2 et E1=Z2, then Pins.h : pins  E0 assigned to
Y2, pins of E1 assigned to Z2
#define Y2_STEP_PIN                        P2_13
#define Y2_DIR_PIN                         P0_11
#define Y2_ENABLE_PIN                      P2_12
#ifndef Y2_CS_PIN
  #define Y2_CS_PIN                        P1_04
#endif

// E1 becomes Z2
// Lowrider if EXTRUDERS = 0 , so marlin will E0=>X1 et E1=>Y2 but
Lowrider2 I want E0=Y2 et E1=Z2, then Pins.h : pins  E0 assigned to
Y2, pins of E1 assigned to Z2
#define Z2_STEP_PIN                        P1_15
#define Z2_DIR_PIN                         P1_14
#define Z2_ENABLE_PIN                      P1_16
#ifndef Z2_CS_PIN
  #define Z2_CS_PIN                        P1_01
#endif

tca72, 2021-03-02:
Hello

Would you mind share with me your configuration files with 1.4 turbo with
Dual EndStop, as right now mine as an issue, the YMax (PWRDET), ZMAX
(E1DET) are not producing any change.

best regards

EynErgy, 2021-03-02:
Hi, no problem.
Just take it from my repo.
I made some changes to prepare work for your config but needs inputs to finish it.

tca72, 2021-03-02:
hello

what kind of inputs?

EynErgy, 2021-03-02:
Hi,
Like the content of your tmc5160 file 
And what is needed for StepperOnline ( a new accessory file for it?)

tca72, 2021-03-02:
Hello

There is no specific file for the tmc5160, just some adaptation as
described in this email I sent yesterday.
But I've seen the configuration that you prepare for the TMC5160, and there
is an error in accessories/tmc5160
SQUARE_WAVE_STEPPING must be disable.

opt_enable \
MONITOR_DRIVER_STATUS \
SQUARE_WAVE_STEPPING \
TMC_DEBUG

 also, please note that nowadays the dual endstop is not working on
SKRV1.4, they are producing no result for PWRDET and E1DET. I still
working to find out a solution, but I suspect that this is not feasible as
I made so much changes and try, that nothing changes.
let's see.

EynErgy, 2021-03-03:
Hello,

Ok I made the changes for square wave and temperature reporting.
Could you please try this build and report your issues?
https://drive.google.com/file/d/1kWW1Jvr4O89q5nsATg4Nst3t8bgH_mqz/view?usp=sharing

Also to be sure if not done yet could you please try all your end stops on Xmin with the command M119?
Just to check if your switches works

EynErgy, 2021-03-05:
No news?

tca72, 2021-03-05:
Hello

I reach to get endstop working for SKRV1.4, but not as I expected.

But I've got an interrogation that perhaps you could answer.

I define 2 endstop on Y and two endstop on Z.  So on lowrider we have 2
stepper motors on Y and 2 steppers motors on Z.

my objective was to get  the first Y (YMin) endstop stopping 1 stepper
motor Y, and the second Y (Ymax) endtop stopping the second motor (Y2).
But nowadays both endstop are stopping both stepper motors. I mean Endstop
(YMin ie E1DET) is stopping the 2 Y motors, and Ymax (PWRDET) endstop is
stopping the 2 Y motors.
Did you got same behaviour on your SKR pro board.

BR

tca72, 2021-03-05:
Hello

I review the code.

In this configuration-adv.h, you did not change the RSENSE to 0.075 ( this
is mandatory for BTT TMC5160) .
from line 2246 and next for all steppers TMC5160 defined in configuration.h.

I notice also that you did not remap in pins_BTT_SKR_V1_4.h Pins of stepper
from Y2 to E0 and Z2 to E1, any reason of that ?
(of course Z2 could use a single stepper driver connected to ZMA, ZMB), but
this is not my case.

so right now I did not test this version.

best regards

EynErgy, 2021-03-05:
Hello,

On my MPCNC each end stop switch is stopping his corresponding stepper.
I changed the value for RSENSE.
For the pins it is correct like that, need to be checked but it must be X Y Z Y2 Z2 for the steppers (instead of X Y Z E0 E1)

Here is a new build: https://drive.google.com/file/d/16ax-6VaqFicZmKrpXk8Le3xBUsMXzjkM/view?usp=sharing

Tell me if this one is ok

tca72, 2021-03-05:
Hello

May I ask you also to zip me your skr marlin files and folders to compare
with mine and try to find out why I got this issue with dual endstop
stopping 2 motors instead of one per endstop.

Regards

tca72, 2021-03-06:
Hello

I am not able to compile the new build is there any specific steps to
prepare ?
I got the BTT SKR V1.4 marlin folders and I replace existing marlin folder
but yours.

jeffeb3, 2021-03-06:
Since this is not related to the builds in V1/MarlinBuilder, can we move this conversation somewhere else, please? The V1 forums would ve fine, or an open issue in the fork.

closed 2021-03-06 by jeffeb3

tca72, 2021-03-06:
Hello
Some changes in the configuration_adv:
for TMC5160, and all TMC steppers drivers with SPI mode, you have to enable
this: (  otherwise it compile but when the board is started, you have the
message ALL HIGH, and board halted)
  #define TMC_USE_SW_SPI

tca72, 2021-03-06:
Hello

Some other changes in configuration_adv.h : set to :
  #define CHOPPER_TIMING CHOPPER_DEFAULT_24V

EynErgy, 2021-03-08:
Hi tca72,

I made the changes, please fill an issue on my forked repository. We will continue there