V1EngineeringInc/MarlinBuilder #41: Print bed is not correct

3DFans, 2020-11-01:
Hey, I downloaded the latest firmware to my printer with the MK8 extruder. However, the dimensions entered are not correct, because each time it is pressed, it moves to the right and walks forward. The same game at the MBL. Is there a solution ? In addition, he does not print the test cube in 20mm but in 25mm. greetings

V1EngineeringInc, 2020-11-01:
For what type of 3D printer?

The direction can be changed in the firmware, or by flipping the stepper plugs. As for the print size it should like you are using different pulleys than the recommended MP3DP 16T, adjust the steps per mm.

3DFans, 2020-11-01:
Hey, I'm using the MP3DP with the MK8 extruder. The print bed is 220x220 and on the left it should be about 1.5cm to the right and 1.5cm to the left on the left. The printer was originally assembled from the parts from the shop. Can you help me with that

jeffeb3, 2020-11-01:
Can you post some pictures and details with the issue in the forums?

1. This space is really about changes needed to the firmware.
2. There are a bunch of users in the forums that can help.
3. The forums are set up to better post details like photos and bits of code.

V1EngineeringInc, 2020-11-23:
If this is still a problem it sounds like a configuration issue more than the stock firmware (mine are running good).

I will close this out and see you in the forums if there are any further issues.

closed 2020-11-23 by V1EngineeringInc