V1EngineeringInc/MarlinBuilder #44: Archim 1 microstepping not being set properly.

V1EngineeringInc, 2020-11-02:
The discussion, https://forum.v1engineering.com/t/steppers-moving-at-different-rates/21925

The current fix, 

> M350 S16
> M351 S3 X1 Y1 Z1 E1 B1
> M500

kubark42, 2021-03-29:
I might see what's going on here:

https://github.com/MarlinFirmware/Marlin/blob/2.0.7.2/Marlin/src/module/stepper.cpp#L3403 shows that it needs `HAS_MICROSTEP32` to be declared.

https://github.com/MarlinFirmware/Marlin/blob/2.0.7.2/Marlin/src/inc/Conditionals_post.h#L2035 needs `MICROSTEP32` to be declared in order to declare `HAS_MICROSTEP32`.

A little higher up, https://github.com/MarlinFirmware/Marlin/blob/2.0.7.2/Marlin/src/inc/Conditionals_post.h#L2016 shows that only for the Alligator board is the `MICROSTEP32` value declared.

https://github.com/V1EngineeringInc/MarlinBuilder/blob/510/src/configs/boards/archim1 directs there to be a MICROSTEP32 declared, but this exported setting is not making into `Configuration_adv.h`.

```
/**
 * Custom Microstepping
 * Override as-needed for your setup. Up to 3 MS pins are supported.
 */
//#define MICROSTEP1 LOW,LOW,LOW
//#define MICROSTEP2 HIGH,LOW,LOW
//#define MICROSTEP4 LOW,HIGH,LOW
//#define MICROSTEP8 HIGH,HIGH,LOW
//#define MICROSTEP16 LOW,LOW,HIGH
//#define MICROSTEP32 HIGH,LOW,HIGH

// Microstep settings (Requires a board with pins named X_MS1, X_MS2, etc.)
#define MICROSTEP_MODES { 32, 32, 32, 32, 32, 32 }
```

Perhaps because it is written `opt_enable ` instead of `opt_set`? I cannot find the definition of `opt_enable` and `opt_set`, it seems to come in on the builder and so I'm into a dead end here.

However, I can confirm that the problem is resolved by replacing `#define MICROSTEP_MODES { 32, 32, 32, 32, 32, 32 }` with`#define MICROSTEP_MODES { 16, 16, 16, 16, 16, 16 }` in `Configuration_adv.h`. When I send the M350 g-code command, I receive `MS1|2|3 Pins X:110 Y:110 Z:110 E0:110 E1:110`.

kubark42, 2021-03-29:
https://github.com/V1EngineeringInc/MarlinBuilder/blob/510/src/configs/boards/archim1 directs there to be a MICROSTEP32 declared, and this was added in https://github.com/V1EngineeringInc/MarlinBuilder/commit/fd4b8307d33cd6d67639233992dba969ba9f01f1#diff-48db6c243fc1b1270804d825da54f17e4eb427bed2d605fb12438653ffe4e72b. That closes out this issue, correct?

V1EngineeringInc, 2021-03-29:
Yes, thanks.

closed 2021-03-29 by V1EngineeringInc