V1EngineeringInc/MarlinBuilder #46: Need to define a laser pin for all the boards.

V1EngineeringInc, 2020-11-18:
Best to also add a link to the amazon IOT relay in the instructions somewhere. The one that can run off the fan pin...and some instructions, so maybe a separate page in the docs? https://amzn.to/35CFbDP

Need to try and sort out the laser issues.
1-Does M3 work as expected now, no under or over burning? If not M106
2-Define a 5V pin to use with the lasers, preferably next to a ground. ....on most of the boards....Mini, Rambo, SKR.
3-Update either the laser page or the board pages with instructions (probably both).

V1EngineeringInc, 2020-11-19:
M3 seems to be great now!!!

Pins for the SKR pro, Thanks to Tuco, [pretty awesome when the first google results are your own site!](https://forum.v1engineering.com/t/how-to-modify-marlin-to-control-a-laser-with-an-skr-pro/21712)

proof of work...first laser etch in a very long time.
[image]
Lots of settings to test but that was all the bells and whistles in Marlin enabled (Continuous power ect.).

V1EngineeringInc, 2020-11-21:
The continuous power does not seem to work as well as just using over scan. Maybe revisit it later, probably best for vector based etching.

Not sure what is happening but I am getting glitchy burns, and looks like inline commands do not like "M5" so to end a burn you just use S0??

V1EngineeringInc, 2020-11-21:
[image]
Glitchy

jeffeb3, 2020-11-21:
I wonder if there is a transmission issue, like a wart on a 3d print.

V1EngineeringInc, 2020-11-21:
So many things to try, not really sure. I have tried 4 different firmware settings since that post. Without doing inline/or continuous commands it does not burn well so that narrows it down some.

Double checking the trapezoid ramping. see if that is better, worse, same.

V1EngineeringInc, 2020-11-21:
I thought it might have been a commands per second issue. tried a much larger dot size and nothing changed.

jeffeb3, 2020-11-21:
I was thinking, are you running from USB, or the new tft? Maybe run from an old lcd/sd card to eliminate the command pausing.

V1EngineeringInc, 2020-11-21:
Marlin mode on the tft, you think could be it?.....dam. Let me try tomorrow I just shut it down, Good idea, thanks!

jeffeb3, 2020-11-21:
I'm not sure. If it was pausing that would explain it a bit. I don't really understand what you are trying and which configs aren't working (I haven't looked through the setiings myself).

V1EngineeringInc, 2020-11-21:
No that didn't change anything.

Tried higher acc, jerk, max speed this morning.

I have messed with all the laser settings except frequency I don't think it will do anything but I will try it next. 
`#define SPINDLE_LASER_FREQUENCY       2500   // (Hz) Spindle/laser frequency (only on supported HALs: AVR and LPC)`

Odd thing is I just tried to run it in TFT mode and I can not run the files? It sees them but nothing happens when I click them.

V1EngineeringInc, 2020-11-21:
https://github.com/MarlinFirmware/Marlin/pull/20202

Hmmm, still pretty active development going on for M3. Might need to go back to M106

V1EngineeringInc, 2020-11-21:
Okay swapped laser, driver, and wiring. Same issue so the first thing has been ruled out....not the laser itself.

V1EngineeringInc, 2020-11-22:
[image]
So...

Laser, wires, memory cards, files, burning software, m3 m106, speeds, accels, frequencies, no S curve, sCurve, all power ramp types, various ramp steps, screens, screen types, file location on the card. Nothing.

The only thing I have not tried at this point...different board. So now, rambo testing instead of SKR.

V1EngineeringInc, 2020-11-22:
Post 2 had no glitches.....the only thing I can think of at this point is that was done at ~20-30mm/s not the 55-60mm/s the rest were. I really hope it is that easy.

V1EngineeringInc, 2020-11-22:
Eureka!!

Speed...this whole time it was speed.

I will keep testing to see what the limits are.

V1EngineeringInc, 2020-11-23:
SKR is good to go. 8 bit boards up next.

Taking more detailed searchable notes here, https://forum.v1engineering.com/t/lasers-revisted/22532/6

Allted, 2020-12-06:
Glitching seems to be an issue with Classic Jerk so I submitted a Change to go back to Junction deviation. I will test again.

jeffeb3, 2021-02-01:
Is this issue closed?

V1EngineeringInc, 2021-02-01:
yes/no. I have not really updated any of the boards pages or the laser page....and there are a few board left.

closed 2023-03-17 by V1EngineeringInc