V1EngineeringInc/MarlinBuilder #60: more configs

jeffeb3, 2021-01-23:
We need to make some more configs:
- 3DP Rambo?
- CNC Ramps?

V1EngineeringInc, 2021-01-23:
We already have ramps and ramps dual.

jeffeb3, 2021-01-23:
V1CNC Ramps DualLR then? I remember people asking about a specific ramps combo in the forums, but now I can't figure out which one it was.

Today there was a question about a Rambo 3D Printer configuration and they were pointed to the mini rambo. It wasn't for an MP3DP.

jeffeb3, 2021-01-30:
3DP skr pro?

What about the skr mini?

Benush26, 2021-04-22:
Any chance the SKR2 is in the works or should I use something like skr1.3 and hope I can figure out all I need to?

V1EngineeringInc, 2021-04-22:
The best option is just to use a SKR Pro if you are buying a new board. 

There are no plans to support that board in particular, but if you got it (and if it is supported by Marlin fully yet), editing one of our current builds could work. It really depends on how much is actually different. Sometimes they just make the boards better, other times they change pins and processors.

Benush26, 2021-04-22:
Thank you.
I do already have the SKR2.  After looking at the specs, it appears to be the board they should have introduced instead of the 1.3, 1.4 and 1.4 turbo.  Now if it just holds up.
I’m not a Marlin expert, but now that I’m on my second 3D printer I’m comfortable dealing with the configuration, configuration.adv and platformio files.
If I need to get in to the src directory I’ll just keep playing until I find what I need.
Thanks for taking the time to answer my question.
Be well and stay safe,
Ben

Benush26, 2021-05-03:
And fate steps in!  BTT goofed with their SKR2 build and are giving me a full refund which allowed me to get the SKR pro which should be here early next week.  Long before I get the trucks and core printed.
Thanks again for your reply!  Learning to print in FDM was a true growth experience, especially for an old fart like me.  I took apart and continually upgrade an older i3 clone (FolgerTech) which gave me the confidence to build you MPCNC.  As soon as the cnc is completed and tested, I presume that learning to now ‘remove’ material will be equally as challenging.😱😁
Thanks and stay safe and be well,
Ben