V1EngineeringInc/MarlinBuilder #73: Document how to flash firmware via main README.md

kubark42, 2021-03-29:
This should probably just be a link to the relevant docu on v1engineering.com. 

However, when going to https://docs.v1engineering.com/electronics/marlin-firmware/ I don't see information for flashing the `.bin`. I try to keep Platform.io off my computer, in the past it hasn't played nicely with mission critical stuff from Paticle.io. Hopefully there is an easy and lightweight way to flash the binary, rather than install all of VS Code and Platform.io. 

P.S. FWIW, in my experience flashing the compiled binary tends to reduce maintenance and support burden, since the project devs can be certain that everyone is on the same page. In particular, it eliminates the case of hardware failure linked to divergent build settings.

jeffeb3, 2021-03-29:
The skr instructions are here:

https://docs.v1engineering.com/electronics/skrpro/#firmware

I have a bit if a rewrite in a PR that suggests to use xloader for atmega boards (ramps/rambos). It is probably time to look closer at that, since we haven't heard any complaints and the releases page is working.

I think we probably need to point at the docs page for MarlinBuilder releases from here and point to the right place for flashing from there.

kubark42, 2021-03-29:
I have an Archim 1.0a (I updated my original post about 1 minute before your response showed up. My apologies!), so am on the hunt for a SAM3X8E flashing strategy. 

> I think we probably need to point at the docs page for MarlinBuilder releases from here and point to the right place for flashing from there.

I think that makes a ton of sense. Placing docu here will ultimately lead to competing versions, with readers trying to figure out which is fresh and which is stale.

jeffeb3, 2021-03-29:
I don't have an archim, but they seem fickle. I am not sure you will get any shortcut like we have for the other boards. I would use the arduino ide and v507. Marlin broke it in 2.0.7.2 and they don't seem interested in fixing arduino ide.

kubark42, 2021-03-29:
<thread_drift>

I have been reading about V1's trials and travails with this board. As an owner of an Archim 1.0a I hate to say it, but it looks like the hardware should be rapidly deprecated. Supporting buggy hardware which has been abandoned by the upstream manufacturer is a Sisyphean task. I have been less than impressed by Ultimachine and their unstable firmware stack.

</thread_drift>

V1EngineeringInc, 2021-03-29:
Using 507 you still get all the features, so currently it has not really fallen behind yet. I think you can even use the new firmware you just need to manually set the MS pins. The commands for that are simple and in the forums if you really want to go that way.

But you are correct. I do not plan to purchase any more, and I am doing my best to continue support when a new firmware feature comes out, but there really is no need to keep it on the bleeding edge.

kubark42, 2021-03-29:
One way how to directly flash the binary is to use the flashing tools installed with Arduino. A relatively unknown feature in Arduino is the ability to output the console commands used by Arduino IDE. Arduino IDE is actually very simple, and doesn't do any of the compilation itself. It spins all that out to CLI packages.

[image]

So by turning on `Show verbose output during      upload`, I was able to extract the below command:

`~/Library/Arduino15/packages/arduino/tools/bossac/1.7.0/bossac -i -d --port=cu.usbmodem142301 -U true -e -w -v -b ~/Downloads/V1CNC_Archim1-2.0.7.2-src/firmware.bin  -R`

I've built entire CI software toolchains on top this approach, so I do trust it to work consistently. I've spent 15 minutes right now playing around, flashing the `.bin` which are included in https://github.com/V1EngineeringInc/MarlinBuilder/releases.

The only things a user needs to set are the location of bossac (although that could be gotten independently of Arduino), the port (in this case, `cu.usbmodem142301`), and the path to the binary.

V1EngineeringInc, 2021-03-29:
That works for now (maybe, long filename issues still?) with about as much effort as using platform io. Marlin has dropped support for Arduino so it is best to just follow their lead to P IO.

kubark42, 2021-03-29:
> That works for now (maybe, long filename issues still?) with about as much effort as using platform io.

The CLI solution does not require platform.io. I can see a lot of clients being overwhelmed by Platform.io, esp. when it breaks other Atom-based integrated solutions. The client-side burden of a CLI approach is a copy/pasta command, with very little opportunity for error. With a little bit of scripting, this turns into a cross-platform "double-click to flash" solution. This would be useful, no?

My feeling is that `bossac` will  continue to work outside of Marlin's support of Arduino. The compilation environment is completely decoupled from the flashing tools. Your CI system seems to be doing a great job of automatically generating the binaries, so unless there are custom setting changes I can only see risk in having users recompile.

jeffeb3, 2021-03-29:
FYI, it does not work with atom anymore. Only visual studio code (which is very cross platform). 

At any rate, I am not excited about doing extra for the archim. Unfortunately, it was an experiment that hasn't worked out. It has received a lot of extra attention already. 

Perhaps this info is a better fit for the forums, where rhe remaining archim users hang out and can find it.

kubark42, 2021-03-29:
No debate there! I had thought that Archim was the outlier and everything else was Arduino based. the info is here if you guys need it for any other future Arduino-based platforms.

> FYI, it does not work with atom anymore.

Yeah, I mistyped that. Platform.io and Particle.io both were using VS Code back around 2018, and therein lay conflicts. They didn't play nicely together, sort of like an IDE version of conflicting environmental variable. This seems to be a frequent refrain when reskinning another tool to create a bespoke IDE, so I never really dove into it.

jeffeb3, 2021-03-29:
Bossac isn't used by the atmega or the skr boards. The skr just loads it from microsd card. The atmegas can use xloader. They use avrdude in arduino.

thijstriemstra, 2023-03-11:
> I try to keep Platform.io off my computer, in the past it hasn't played nicely with mission critical stuff from Paticle.io. Hopefully there is an easy and lightweight way to flash the binary, rather than install all of VS Code and Platform.io.

I love platformio, and to flash firmware I use:

```bash
export port=/dev/ttyUSB0

platformio run --upload-port ${port} --monitor-port ${port} --target upload -e mega2560
```

Adding this to the readme should fix this ticket imo.

closed 2023-03-17 by V1EngineeringInc