V1EngineeringInc/MarlinBuilder #75: Clarify if `dual` refers to endstop or stepper motor drivers

kubark42, 2021-03-29:
It is a little unclear to me whether the `dual` badged firmwares are for "dual endstop" support or "dual stepper motor driver" support. This might be nice to have in the `README.md`, or perhaps as boilerplate for release descriptions.

jeffeb3, 2021-03-29:
https://docs.v1engineering.com/electronics/marlin-releases/#dual-or-series-stepper-wiring

This repo should be light on info, and point to the docs. 

FWIW, there hasn't ever been a dual motor without endstops configuration.

closed 2021-03-29 by jeffeb3

kubark42, 2021-03-29:
> This repo should be light on info, and point to the docs.

Agreed wholeheartedly. I guess this ties into #73. Maybe put https://docs.v1engineering.com/electronics/marlin-releases/#decoding-the-config-names into the main `README.md`? It's a well-documented naming convention with good thought put into it, I just hadn't found it in my searches. Thanks!