V1EngineeringInc/MarlinBuilder #76: Rambo board cnc dual is setup for dual z instead of dual x

mmerkel22, 2021-04-01:
I downloaded the Rambo dual cnc firmware on 3/31 and it compiled and loaded fine but I found that it was setup for dual y and z drivers. I changed it to dual x and y and the motion is fine now. Just wanted to let you know. Keep up the great work. Thanks.

V1EngineeringInc, 2021-04-01:
If it had dualLR in the title that is for the Lowrider. Can you check your zip file name please.

mmerkel22, 2021-04-01:
It was the V1_CNC_Dual

jeffeb3, 2021-04-01:
Can you post the exact filename? I will gladly trace this issue, but the thing that makes the most sense is that you accidentally grabbed the LR version.

V1CNC_Ramps_Dual
is different than 
V1CNC_Ramsp_DualLR

mmerkel22, 2021-04-01:
Yes you are correct. My bad. I downloaded both but opened the low rider in vs code.

jeffeb3, 2021-04-01:
Thank you for checking.

closed 2021-04-01 by jeffeb3