V1EngineeringInc/MarlinBuilder #78: Marking the releases as `...-src.zip` is confusing to people who are looking for `.bin` files

DanRoscigno, 2021-04-17:
I think it would be nice to drop the `.src` from the release filenames, or clearly state in the release notes that the src.zip files include both the Marlin source and precompiled .bin files.

For example, the `MPCNC Primo with SKR Pro board using Dual Endstops` release file would be:
`V1CNC_SkrPro_Dual_2209-2.0.7.2.zip`

and include 
- `firmware.bin`
- `V1CNC_SkrPro_Dual_2209-2.0.7.2-src.zip`

V1EngineeringInc, 2021-04-17:
I'm okay with that.

It is as simple as removing it from both lines here, https://github.com/V1EngineeringInc/MarlinBuilder/blob/main/src/core/zip-marlin?

jeffeb3, 2021-04-17:
I think you want to leave the -src on line 39. But remove the -src from line 41.

V1EngineeringInc, 2021-05-06:
Looks to be done, can't be 100% certain until we do another release though.

Thanks Dan

closed 2021-05-06 by V1EngineeringInc