V1EngineeringInc/MarlinBuilder #84: V1CNC_SkrTurbo_8825-2-3  Binary File via SD

CoefficientofSquish, 2021-04-29:
I've been compiling and flashing firmware on my SKR 1.4 Turbo for a couple days and just now when I tried to put the precompiled "V1CNC_SkrTurbo_8825-2-3" binary file on the SD and flash it I'm only getting M105 over and over in the terminal.  I don't know if this necessarily a problem with the file because I've had this happen before when trying to upload a binary via SD card on SKR 1.4 Turbo so it could be either something I'm doing wrong or something about the boards but everything else I've flashed that way has worked.  I have also noticed that both of the SKR 1.4 Turbo's that I've had could never be recognized by my computer when connected by USB, or even once they could I could still never get PlatformIO to flash via USB so it could have something to do with that also.  

It definitely tried to flash though, because I tried it a second time to make sure the bin file was on the SD and both times the FIRMWARE.BIN was gone and replaced by a new FIRMWARE.CUR, only it's partially bricking the board for some reason.

V1EngineeringInc, 2021-04-29:
I don't have a skr turbo to test with, you might want to post in the forums to see if anyone with some experience can help out more. At this point we can't be sure it is a firmware issue without a second confirmation.

CoefficientofSquish, 2021-04-30:
Ok.  It definitely wasn't a hard brick, when I compiled myself it flashed fine so I'm not sure what happened.

jeffeb3, 2021-04-30:
Agreed, this should be in the forums.

closed 2021-04-30 by jeffeb3