V1EngineeringInc/MarlinBuilder #93: Updates for next release

jeffeb3, 2021-07-22:
Some things I've seen in the forums that I'd like to investigate:

- Bump  version to 2.0.9.x
- 2.0.8.2 has added a 3rd serial port for the skr boards. This should allow wifi dongles as well as the usb and tft. Needs some testing.
- Robert suggests we add emergency parsing. I'm not sure the consequences yet, but I think it is worth it for the users that need it.

Allted, 2021-08-30:
I saw it coming, arcs got revamped and causing the builds to fail. I will try to test the new settings asap.

InoxDragoon, 2022-01-25:
Can Host Action Commands be added for those of us who use OctoPrint? it's part of the 2.0.9.2+ marlin builds.

V1EngineeringInc, 2022-01-26:
Are any of the other options needed under that define?

If enabled, will the firmware still fit on a Ramps board?

V1EngineeringInc, 2022-01-27:
I think this needs to be discussed in the forums. No one has mentioned it before, and I am not sure if many others would benefit from it.

 I would need this to be at least complied to see if it fits on a Ramps board we are currently still supporting them.

V1EngineeringInc, 2022-02-09:
Okay, emergency parsing and host_prompt_commands will get a shot soon. 

https://github.com/MarlinFirmware/Marlin/blob/0a24f858f3009b358c3c0841654521146949261a/Marlin/Configuration_adv.h#L3953 Any idea where to look into more information about the other options? Does octoprint have a guide?

The skr might have an issue, https://community.octoprint.org/t/how-do-i-make-marlin-send-out-action-commands/10522/2

Foosel only seems to list the main option, so I guess we start there, https://community.octoprint.org/t/octoprint-tells-me-my-firmware-lacks-support-for-host-action-commands-what-does-this-mean/34588

This needs to be done on a day I can flash and test at least the Rambo and SKR Pro versions.

closed 2023-03-11 by V1EngineeringInc