V1EngineeringInc/MarlinBuilder #134: Enable Laser without Safety timeout for Laser Focusing

leitenky, 2022-10-13:
Currently I have a Endurance Laser with power from Fan port and PWM coming from Pin45. With the latest V515 firmware there is no way I can Power on the laser to allow focus for more then 1 second.  I Can update LASER_SAFETY_TIMEOUT_MS in the firmware to use a longer time, but i would prefer to keep the safety untouched and have a gcode command / macro that could temporarily disable the timeout. 

Previous Controls for laser focus -v513
M106 S255 ; Power on laser
M3 S30 ; fire laser with low power to focus
*focus the laser manually*
M5 ; Stop laser
M107 ; Power down the laser

Current version -v515
M106 S255 ; Power on laser
M3 I S0 ; without this after the first laser fire it will not fire again
M3 S30 ; fire laser with low power to focus
*focus the laser manually* while resending M3 S30 multiple times
M05 ; Stop laser
M107 ; Power down the laser

V1EngineeringInc, 2022-10-13:
You should not need both M3 I and M106. Does either one work better by itself?

I will need to look into this a bit, in the mean time I have a focusing script that fires a test pattern, or there is one built into light burn as well I believe.

Please open up a forum topic so we can discuss this further with some of the crew that uses the laser more than I do.

leitenky, 2022-10-13:
I take my power from a fan port for the laser so M106 S255 powers up the controller / cooling fan on the laser. I didn't direct wire it to the 12v power as there was no way to turn off the laser fan when the endmill is in use.  Then if I don't use the M3 I S0 the laser won't "fire"

V1EngineeringInc, 2022-10-13:
Okay. I would still like to discuss this in the forums. Please open a topic and tag me in it "@vicious1". Just need to verify there is no other repercussions of this.

closed 2022-10-15 by V1EngineeringInc