V1EngineeringInc/MarlinBuilder #141: Fix or remove arch builder

thijstriemstra, 2023-03-17:
For years the ci build status of this project has been failed, solely because of the Archim board/architecture it seems. I would suggest to allow it to fail and not set the overall build status of the project to failed. Otherwise it seems this project will never have a proper ci build (at first glance that was my impression at least).

V1EngineeringInc, 2023-03-17:
It only fails from the dual versions of the archim. That said, I almost took it out last week. I really don't mind taking it out. I will do it next time I make some changes, or if you want to do it sooner I will merge it.

https://github.com/V1EngineeringInc/MarlinBuilder/actions

thijstriemstra, 2023-03-19:
Sounds good @V1EngineeringInc If you need someone to review that future PR, let me know and I'll take a look.

PR #147 (2023-06-27): ci: remove failing archim builder
fixes #141
commit c09b2d58 (2023-06-27): ci: remove failing archim builder

fixes #141
closed 2023-06-27 by V1EngineeringInc